pimoroni/enviro #177: How to disable Led  when it wakes up from rtc or gets power?

notsource, 2023-05-19:
Hi, the white led is turning on for a sec or two when the device gets power or wakes up from rtc. Is it possible to disable the led when it wakes of from RTC? And even if not can someone explain please. I came to the conlusion this is build in the hardware itself?

I'm running v0.0.2 

Thank you in adance

sjefferson99, 2023-05-20:
I recommend you install at least v0.0.9 as there was much done since v0.0.2 that makes a lot of things work correctly. Ideally you would then copy down the files from the main branch and overwrite using Thonny as I believe there are more useful additions not yet captured in a release.

The pulsing white light is designed to show the board is working, but it is defined in software, specifically the function to generate it is in enviro/__init__.py and it is called on startup in the same file.

If you are using the latest code from the main branch it's on line 495 ("pulse_activity_led(0.5)") and I suspect you could simply comment that line out to stop the white light showing. The red warning light would still show, but that should be exceptional and is worth retaining unless you're after a stealth weather station.

You could perhaps put a new option in the config file using the template and defaults files along the lines of disable_white_led = True and then wrap line 495 in an appropriate if statement to make that more accessible.

notsource, 2023-05-21:
hi jeff, thank you for your reply. I was concerned about the recent bugs with battery etc. I'm not sure on which version im really either. I commented the enviro.startup() out which includes the pulse_activity_led() function but the light still showed up. Its a steady light for 1-2 sec when the device turned on on battery and on USB.

I managed to fix it when i first flashed it with with a basic micropython rp2 port pico w firmware. Then I flashed it with the latest pimorino pimoroni-picow-v1.20.2-micropython.uf2. With the same files on the filesystem. Now i can toggle the pulse activity led when I really needed and nothing else. The warn led lights up only when the battery are to low(need to confirm that later, as the battery drained from testing :D. For notification I will use the buzzer to know whats going on. 

cheers

notsource, 2023-05-28:
This Issue can be closed. I used some plado like gum to isolate the light. Can easily be removed if I want to check the status.
Went back to 0.0.2 firmware and the is wifi stable again.

closed 2023-05-28 by notsource