pimoroni/enviro #180: Enviro Urban blocks my website link

PMSauve, 2023-06-14:
The Enviro Urban project that I created on 2023-06-04 must be deleted :
- it has blocked an important link on my website directing to my exhibition of artwork in a museum. 
So on date of 2023-06-14, in my Adafruit account, I deleted my project in the Feeds.
But now, I still am unable to get the link on my website to work.
Can someone please help?

guru, 2023-06-14:
@PMSauve Where did you create this project? I'll need more details so we can assess what needs to change with the code here on GitHub.

PMSauve, 2023-06-14:
That is so kind, I am in meeting for one hour, will send details in about 2 hoirs.thank you,Paulette                                                                                                                                                                        Sent from my BlackBerry - the most secure mobile device - via the Rogers Network                                                                                                                                                                            From: ***@***.***: June 14, 2023 13:47To: ***@***.***: ***@***.***: ***@***.***; ***@***.***: Re: [pimoroni/enviro] Enviro Urban blocks my website link (Issue #180)  
@PMSauve Where did you create this project? I'll need more details so we can assess what needs to change with the code here on GitHub.

PMSauve, 2023-06-15:
Hello guru, I will try to give the details that show how I discovered that Enviro Urban Setup was causing a problem.
I am able to open links to any other website, but not the one that is shown here.
I asked a few friends and they have no probem opening the following link. 
But for me, the following link which was working fine for several weeks, stopped working just a few days ago. 
The problematic link is to a museum exhibiting my artwork and it was working fine on my website is for several weeks:
https://mbas.qc.ca/paulette-marie-sauve-vibrations-sur-fibres-23/

I used 3 different browsers and get the same ERROR MESSAGE:

"This site can’t be reached
mbas.qc.ca took too long to respond.
Try:
Checking the connection
Checking the proxy and the firewall
Running Windows Network Diagnostics
ERR_CONNECTION_TIMED_OUT"
So I decided to run Windows Network Diagnostics.  
Can you kindly tell me if I can safely put the Diagnostics details into GitHub?
Thank you, Paulette   :)

PMSauve, 2023-06-15:
Please forgive me, I should add that this project came from here:
Pico Enviro Urban : a quick start guide on Github:

https://github.com/pimoroni/enviro/blob/main/documentation/boards/enviro-urban.md

Thank you, Paulette  :)

closed 2023-06-15 by guru

PMSauve, 2023-06-15:
Thank you, I will contact them about this issue. 
This problem is perhaps related to the day (June 4, 2023) that I connected the Enviro Urban device on my laptop.
What I intended to do with the Enviro Urban project was to gather data for my Biodiversity project.
This data was to be placed next to a hand woven tapestry of a Great Blue Heron. 
Here is the page on my website (website is English and French) that shows the link that I am unable to open:

https://www.paulettesauve.com/product/vibrations-on-fibers-23-vibrations-sur-fibres-23/

Thank you, Paulette    :)