pimoroni/enviro #182: Enviro Urban conflict with museum website

PMSauve, 2023-06-23:
On June 14, one of the GitHub members very kindly suggested how to fix my problem with the Pico Enviro Urban.  Since then I have cleared all the data from the flash memory on my Enviro board, and I had some kind help from a Pimoroni agent, but the Museum website link to my website has me very worried. The Museum is currently giving wonderful promotion in the media for my exhibition which includes a tapestry that should have had an Enviro Urban connection for a biodiversity project. In first week of June, after putting the project into my io.Adafruit account, and my GitHub account, and then when I tried to announce this project on my own website I suddenly lost my internet connections to the Museum website. So I removed the project from my io.Adafruit account, and my GitHub account.  However, the BELL internet server in my residence still won't allow me to connect to the museum website.  BELL has a list of suggestions that they give to every client who needs help, and none of those suggestions fixed the problem.  Can any member here please help?

Gadgetoid, 2023-06-23:
Based on your previous messages at https://github.com/pimoroni/enviro/issues/180 I conducted some local tests to try and replicate your problem and trigger some sort of blocking action against my browser/computer. I was unsuccessful.

I don't like to speak in absolutes. But I don't believe Enviro is the cause of your problem.

This means, I'm afraid, that there is nothing we can do to help- we have no control over your ISP, or of the museums servers, nor any technical insight into how they might operate. I can only guess at the cause of your problem and offer suggestions and unhelpful platitudes.

It would be my recommendation to find the person at mbas.qc.ca responsible for their website, and relay your problem to them. I am only making assumptions here, but I assume they are  - with the addition of your ISP, BELL - one of the only two organizations capable of solving your problem.

This means you should find the most technical (or senior) sounding person on this page - https://mbas.qc.ca/equipe/ - and find out if they can ask their website hosting provider whether their server has blocked your specific IP address. Your "IP address" should be the number you see when visiting https://api.seeip.org/ on the affected computer.

PMSauve, 2023-06-23:
Thank you Gadgetoid, I appeciate your help!  :)   I will follow your suggestions.  
Best regards, Paulette