pimoroni/enviro #195: enviro grow takes readings only after manually running main.py

VGONUL, 2023-08-10:
I have set up InfluxDB and Grana Dashboard, but everytime Do I need to "reset the board" to get data ? or Am I missing something ?

helgibbons, 2023-08-10:
No, it should wake up and take readings automatically according to the `reading frequency` that's set in `config.py`. You can press the poke button if you want to wake it up and take a reading straight away though.

VGONUL, 2023-08-10:
I tried again and again, unfortunately same result, When I ran the main.py , I could import the data and see it,but removing USB and reset the board,then it does not transfer the data to the database.

I checked config.py file.I set even 1 for both parameters.

helgibbons, 2023-08-11:
Are you pressing the front poke button to wake it up and trigger a reading after pressing reset?

Do you get any LED activity when you press the poke button?

VGONUL, 2023-08-12:
This is the log file when I connected with USB,and ran the main.py,it worked well but after unplug usb and  trying to get data automatically ,it does not send any data to InfluxDB.

I just recognized two things, This is my http://192.168.1.21:8086, IP Address for InfluxDB.and when I ran 
as you can see below (in the code , It connects a different IP address**2023-08-12 03:35:12 [info     /  97kB]   - ip address:  192.168.1.25**)

But I see this Datapoint in Grafana

[image]

and also System Time as below is different than below timestamp data, Real Time is 05:35:06 but in the Log file it is 03:35:06.

 -  --  ---- -----=--==--===  hey enviro, let's go!  ===--==--=----- ----  --  -     

2023-08-12 03:34:56 [debug    / 123kB] > performing startup
2023-08-12 03:34:56 [info     / 117kB]   - wake reason: button
2023-08-12 03:34:56 [debug    / 115kB]   - turn on activity led
2023-08-12 03:34:56 [debug    / 112kB] > 102 blocks free out of 212
2023-08-12 03:34:56 [debug    / 110kB] > taking new reading
2023-08-12 03:34:56 [info     / 106kB]   - seconds since last reading: 212
2023-08-12 03:34:59 [info     / 121kB] > sensor A below moisture target 50 (currently at 0).
2023-08-12 03:34:59 [info     / 119kB]   - running pump A for 2.0 second(s)
2023-08-12 03:35:01 [info     / 109kB] > sensor B below moisture target 50 (currently at 0).
2023-08-12 03:35:02 [info     / 107kB]   - running pump B for 2.0 second(s)
2023-08-12 03:35:04 [info     /  97kB] > sensor C below moisture target 50 (currently at 0).
2023-08-12 03:35:04 [info     /  95kB]   - running pump C for 2.0 second(s)
2023-08-12 03:35:06 [debug    /  85kB] > caching reading for upload
2023-08-12 03:35:06 [info     /  80kB] > 1 cache file(s) need uploading
2023-08-12 03:35:06 [info     /  78kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2023-08-12 03:35:12 [warning  / 100kB]   - took 6 seconds to connect to wifi
**2023-08-12 03:35:12 [info     /  97kB]   - ip address:  192.168.1.25**
2023-08-12 03:35:13 [info     /  91kB] > uploading cached readings to Influxdb bucket: home
**_2023-08-12 03:35:13 [info     / 110kB]   **- uploaded 2023-08-12T03_35_06Z.json**_**
2023-08-12 03:35:13 [info     / 108kB] > going to sleep
2023-08-12 03:35:13 [debug    / 106kB]   - clearing and disabling previous alarm
2023-08-12 03:35:13 [info     / 104kB]   - setting alarm to wake at 03:40am
2023-08-12 03:35:13 [info     / 102kB]   - shutting down
2023-08-12 03:35:13 [debug    / 100kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead

VGONUL, 2023-08-12:
I tried one more time to test my findings 
Setup Config.py is as below.
# enviro config file

# you may edit this file by hand but if you enter provisioning mode
# then the file will be automatically overwritten with new details

provisioned = True

# enter a nickname for this board
nickname = 'plant'

# network access details
wifi_ssid = 'NETIASPOT-2.4GHz-B31285'
wifi_password = 'xxxxxxxxx'

# how often to wake up and take a reading (in minutes)
reading_frequency = 5

# where to upload to ("http", "mqtt", "adafruit_io", "influxdb")
destination = 'influxdb'

# how often to upload data (number of cached readings)
upload_frequency = 1

# web hook settings
custom_http_url = ''
custom_http_username = ''
custom_http_password = ''

# mqtt broker settings
mqtt_broker_address = ''
mqtt_broker_username = ''
mqtt_broker_password = ''

# adafruit ui settings
adafruit_io_username = ''
adafruit_io_key = ''

# influxdb settings
influxdb_org = 'xxxxxxx'
influxdb_url = 'http://192.168.1.21:8086'
influxdb_token = ''
influxdb_bucket = 'home'

# grow specific settings
auto_water = True
moisture_target_a = 50
moisture_target_b = 50
moisture_target_c = 50
BEFORE------------------------------------------------------------------------------------------
[image]
Time is 06.11
AFTER--------------------------------------------------------------------------------------------
means when I ran main.py

logfile.txt

    -  --  ---- -----=--==--===  hey enviro, let's go!  ===--==--=----- ----  --  -     

2023-08-12 04:12:45 [debug    / 119kB] > performing startup
2023-08-12 04:12:45 [info     / 113kB]   - wake reason: button
2023-08-12 04:12:45 [debug    / 111kB]   - turn on activity led
2023-08-12 04:12:45 [debug    / 109kB] > 102 blocks free out of 212
2023-08-12 04:12:45 [debug    / 107kB] > taking new reading
2023-08-12 04:12:45 [info     / 122kB]   - seconds since last reading: 534
2023-08-12 04:12:48 [info     / 108kB] > sensor A below moisture target 50 (currently at 0).
2023-08-12 04:12:48 [info     / 106kB]   - running pump A for 2.0 second(s)
2023-08-12 04:12:50 [info     /  97kB] > sensor B below moisture target 50 (currently at 0).
2023-08-12 04:12:50 [info     /  94kB]   - running pump B for 2.0 second(s)
2023-08-12 04:12:53 [info     /  85kB] > sensor C below moisture target 50 (currently at 0).
2023-08-12 04:12:53 [info     /  83kB]   - running pump C for 2.0 second(s)
2023-08-12 04:12:55 [debug    / 116kB] > caching reading for upload
2023-08-12 04:12:55 [info     / 112kB] > 1 cache file(s) need uploading
2023-08-12 04:12:55 [info     / 109kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
**2023-08-12 04:12:59 [info     /  92kB]   - ip address:  192.168.1.25**
2023-08-12 04:12:59 [info     /  86kB] > uploading cached readings to Influxdb bucket: home
**2023-08-12 04:13:00 [info     / 107kB]   - uploaded 2023-08-12T04_12_55Z.json**
2023-08-12 04:13:00 [info     / 105kB] > going to sleep
2023-08-12 04:13:00 [debug    / 103kB]   - clearing and disabling previous alarm
2023-08-12 04:13:00 [info     / 101kB]   - setting alarm to wake at 04:15am
2023-08-12 04:13:00 [info     /  99kB]   - shutting down
2023-08-12 04:13:00 [debug    /  97kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead

AFTER----------------------------------------------------------Screenshot

[image]

sjefferson99, 2023-08-12:
Hi @VGONUL 

First, it probably didn't show in your editor, but the config file copy paste has taken the comments to be headings in github markdown. If you encase the config file and any other code in three backticks (\`\`\`) top and bottom (\`\`\`) and check the preview tab that will make the post much clearer to read.

The IP address you are seeing in the log (192.168.1.25) is the IP address allocated to your enviro from the router DHCP server, so you know what IP address to look out for on the network. Given this and your data point in the Grafana screenshot, I suspect the configuration is correct and it is likely that the data point in the screenshot means the other data is there from that upload.

I assume you can see the other data points by examining the raw data in Influx and adjusting the grafana query accordingly as your issue headline is that no data is received at all once on battery. 

If that assumption is correct, can you run the board on battery for 10 minutes to ensure we get a 5 minute reading and upload and then post the logs stored on the board after reconnecting. There will likely be some USB power log entries from plugging in, but if you paste the whole file there should be entries showing the attempted influx upload on battery.

VGONUL, 2023-08-15:
hi @sjefferson99 thanks for your remarks, I especially left enviro grow without touching it for last 3 days in order to see if there is a data flow or not but unfortunately I didnt receive any data.
[image]

MrDrem, 2023-08-15:
@VGONUL can you post the new log file from the Grow device please? that's were any errors will show up.

VGONUL, 2023-08-15:
Hi @MrDrem 
you may find as below, I still believe in that ,There is some issue related with time, because In Poland right now time is 15:05 and when I ran main.py , in the log file it shows as 2023-08-15 13:01:11.

"'
2023-08-15 12:37:41 [debug    /  90kB]   - reset
2023-08-15 12:40:53 [debug    / 115kB] > performing startup
2023-08-15 12:40:53 [info     / 122kB]   - wake reason: usb_powered
2023-08-15 12:40:53 [debug    / 120kB]   - turn on activity led
2023-08-15 12:40:53 [debug    / 118kB] > 102 blocks free out of 212
2023-08-15 12:40:53 [debug    / 116kB] > taking new reading
2023-08-15 12:40:53 [info     / 111kB]   - seconds since last reading: 203
2023-08-15 12:40:56 [info     /  98kB] > sensor A below moisture target 50 (currently at 0).
2023-08-15 12:40:56 [info     /  95kB]   - running pump A for 2.0 second(s)
2023-08-15 12:40:58 [info     / 119kB] > sensor B below moisture target 50 (currently at 0).
2023-08-15 12:40:58 [info     / 117kB]   - running pump B for 2.0 second(s)
2023-08-15 12:41:01 [info     / 108kB] > sensor C below moisture target 50 (currently at 0).
2023-08-15 12:41:01 [info     / 105kB]   - running pump C for 2.0 second(s)
2023-08-15 12:41:03 [debug    /  95kB] > caching reading for upload
2023-08-15 12:41:03 [info     /  90kB] > 1 cache file(s) need uploading
2023-08-15 12:41:03 [info     /  88kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2023-08-15 12:41:08 [info     / 114kB]   - ip address:  192.168.1.25
2023-08-15 12:41:09 [info     / 108kB] > uploading cached readings to Influxdb bucket: home
2023-08-15 12:41:09 [info     /  91kB]   - uploaded 2023-08-15T12_41_03Z.json
2023-08-15 12:41:09 [info     /  89kB] > going to sleep
2023-08-15 12:41:09 [debug    /  87kB]   - clearing and disabling previous alarm
2023-08-15 12:41:09 [info     / 120kB]   - setting alarm to wake at 12:45pm
2023-08-15 12:41:09 [info     / 118kB]   - shutting down
2023-08-15 12:41:09 [debug    / 116kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2023-08-15 12:41:10 [debug    / 114kB]   - reset
2023-08-15 12:41:15 [debug    /  95kB] > performing startup
2023-08-15 12:41:15 [info     / 122kB]   - wake reason: button
2023-08-15 12:41:15 [debug    / 120kB]   - turn on activity led
2023-08-15 12:41:15 [debug    / 118kB] > 102 blocks free out of 212
2023-08-15 12:41:15 [debug    / 116kB] > taking new reading
2023-08-15 12:41:15 [info     / 111kB]   - seconds since last reading: 22
2023-08-15 12:41:18 [info     /  98kB] > sensor A below moisture target 50 (currently at 0).
2023-08-15 12:41:18 [info     /  95kB]   - running pump A for 2.0 second(s)
2023-08-15 12:41:20 [info     / 122kB] > sensor B below moisture target 50 (currently at 0).
2023-08-15 12:41:21 [info     / 120kB]   - running pump B for 2.0 second(s)
2023-08-15 12:41:23 [info     / 111kB] > sensor C below moisture target 50 (currently at 0).
2023-08-15 12:41:23 [info     / 108kB]   - running pump C for 2.0 second(s)
2023-08-15 12:41:25 [debug    /  98kB] > caching reading for upload
2023-08-15 12:41:25 [info     /  93kB] > 1 cache file(s) need uploading
2023-08-15 12:41:25 [info     /  91kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2023-08-15 12:41:29 [info     / 122kB]   - ip address:  192.168.1.25
2023-08-15 12:41:29 [info     / 116kB] > uploading cached readings to Influxdb bucket: home
2023-08-15 12:41:30 [info     / 116kB]   - uploaded 2023-08-15T12_41_25Z.json
2023-08-15 12:41:30 [info     / 114kB] > going to sleep
2023-08-15 12:41:30 [debug    / 112kB]   - clearing and disabling previous alarm
2023-08-15 12:41:30 [info     / 110kB]   - setting alarm to wake at 12:45pm
2023-08-15 12:41:30 [info     / 108kB]   - shutting down
2023-08-15 12:41:30 [debug    / 106kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2023-08-15 12:41:33 [debug    / 104kB]   - reset
2000-01-01 00:01:15 [debug    / 115kB] > performing startup
2000-01-01 00:01:15 [info     / 122kB]   - wake reason: usb_powered
2000-01-01 00:01:15 [debug    / 120kB]   - turn on activity led
2000-01-01 00:01:15 [info     / 118kB] > clock not set, synchronise from ntp server
2000-01-01 00:01:39 [debug    / 115kB] > performing startup
2000-01-01 00:01:39 [info     / 122kB]   - wake reason: usb_powered
2000-01-01 00:01:39 [debug    / 120kB]   - turn on activity led
2000-01-01 00:01:39 [info     / 118kB] > clock not set, synchronise from ntp server
2000-01-01 00:01:39 [info     / 116kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2000-01-01 00:01:45 [info     /  95kB]   - ip address:  192.168.1.25
2023-08-15 12:57:35 [info     /  93kB]   - rtc synched
2023-08-15 12:57:35 [debug    /  90kB] > 102 blocks free out of 212
2023-08-15 12:57:35 [debug    / 122kB] > taking new reading
2023-08-15 12:57:35 [info     / 118kB]   - seconds since last reading: 980
2023-08-15 12:57:38 [info     / 104kB] > sensor A below moisture target 50 (currently at 0).
2023-08-15 12:57:38 [info     / 102kB]   - running pump A for 2.0 second(s)
2023-08-15 12:57:40 [info     /  93kB] > sensor B below moisture target 50 (currently at 0).
[... 24 lines cut ...]
**2023-08-15 13:00:55 [debug    / 114kB] > performing startup
2023-08-15 13:00:55 [info     / 122kB]   - wake reason: button
2023-08-15 13:00:55 [debug    / 120kB]   - turn on activity led
2023-08-15 13:00:55 [debug    / 118kB] > 102 blocks free out of 212
2023-08-15 13:00:55 [debug    / 116kB] > taking new reading
[... cut ...]
2023-08-15 13:01:10 [info     /  98kB] > uploading cached readings to Influxdb bucket: home
2023-08-15 13:01:11 [info     / 114kB]   - uploaded 2023-08-15T13_01_05Z.json
2023-08-15 13:01:11 [info     / 112kB] > going to sleep
2023-08-15 13:01:11 [debug    / 110kB]   - clearing and disabling previous alarm
2023-08-15 13:01:11 [info     / 108kB]   - setting alarm to wake at 13:05pm
2023-08-15 13:01:11 [info     / 105kB]   - shutting down
2023-08-15 13:01:11 [debug    / 103kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead**

"'

MrDrem, 2023-08-15:
OK, so it's defaulting to a GMT time server, not to a Polish one, This would need to be tweaked in the code.

looking at your log file:

upload at 12:41:09
```
2023-08-15 12:41:03 [info / 88kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2023-08-15 12:41:08 [info / 114kB] - ip address: 192.168.1.25
2023-08-15 12:41:09 [info / 108kB] > uploading cached readings to Influxdb bucket: home
2023-08-15 12:41:09 [info / 91kB] - uploaded 2023-08-15T12_41_03Z.json
```
It's on USB power, so doesn't sleep, you poked the button at 12:41:15 on the front to get another reading
```
2023-08-15 12:41:09 [debug / 116kB] - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2023-08-15 12:41:10 [debug / 114kB] - reset
2023-08-15 12:41:15 [debug / 95kB] > performing startup
2023-08-15 12:41:15 [info / 122kB] - wake reason: button
```
at 12:41:25 it uploads that reading too
```
2023-08-15 12:41:25 [info / 91kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2023-08-15 12:41:29 [info / 122kB] - ip address: 192.168.1.25
2023-08-15 12:41:29 [info / 116kB] > uploading cached readings to Influxdb bucket: home
2023-08-15 12:41:30 [info / 116kB] - uploaded 2023-08-15T12_41_25Z.json
```
the board then looks like it does a hard reset for some reason, you can tell this from the time stamp dipping back to 0
```
2021-01-01 00:00:06 [info / 88kB] > entering provisioning mode
2021-01-01 00:00:06 [info / 116kB] > auto detecting board type
2021-01-01 00:00:06 [info / 114kB] - grow
2021-01-01 00:00:06 [info / 113kB] > going into access point mode
2021-01-01 00:00:09 [info / 111kB] - 192.168.4.1
2021-01-01 00:00:09 [info / 109kB] > starting dns server...
2021-01-01 00:00:09 [info / 107kB] > starting catch all dns server on port 53
2021-01-01 00:00:10 [info / 105kB] > creating web server...
2021-01-01 00:00:10 [info / 119kB] > waiting for a client to connect
```
then it kicks off another reading and upload
```
2023-08-15 13:01:05 [info / 122kB] > connecting to wifi network 'NETIASPOT-2.4GHz-B31285'
2023-08-15 13:01:10 [info / 105kB] - ip address: 192.168.1.25
2023-08-15 13:01:10 [info / 98kB] > uploading cached readings to Influxdb bucket: home
2023-08-15 13:01:11 [info / 114kB] - uploaded 2023-08-15T13_01_05Z.json
```

If you look in your influxdb instance, you should be able to see the 3 uploads at 12:41:09, 12:41:29 & 13:01:05, I suspect that they may be timestamped slightly earlier, as the json file gets the actual reading time into it, not the upload time, which influx can deal with.

I might try the following in your config.py:

Change your reading frequency to 2, and your upload to 5. This should give you readings every 2 minutes that are uploaded every 10. I've made the changes below for you.

```
reading_frequency = 2
upload_frequency = 5
```

it's clearly connecting correctly to your influx instance correctly. do be aware that Grafana will aggregate readings if you don't match the reading frequency to the data reading frequency. At this point you should be looking at the data in your InfluxDB instance, not in Grafana.

sjefferson99, 2023-08-17:
@VGONUL As per @MrDrem above, when it is connecting at the times in the logs above it is confirming successful send to influx so it's definitely worth confirming they have arrived as a start. I concur, moving directly to influxdb web UI and querying the results in the data explorer function is the easiest way to confirm this. 

In the top right of the data explorer, set the time from local to UTC, that will make it easier to select an appropriate time filter, but also trying going back 24 hours to find the data to zoom into.

The enviro will submit time in UTC and certainly weather stations and sites like weather underground expect the time to be in UTC and the expectation is that you would adjust to a local time zone in your reporting software. Although it would be possible to add timezone adjustment in, it gets complicated quickly if you have to compensate for DST. I have some PRs here that hack at that for the UK, but I highly recommend working in UTC, certainly at first.

Once we have confirmed that the data is being stored correctly on send, it's worth checking what is going on in terms of upload frequency. The attached log has some oddities as per @MrDrem 's analysis and at one point it enters provisioning mode which means either the flag was unset or the whole config file was missing on boot, but then normal provisioned behaviour continued within 2 minutes of the last operation, which is quick if you did reprovision it.

It might be worth flash nuking the board and redownloading the latest firmware and restoring your config file (or reprovisioning as you prefer). But either way setting the suggested frequencies and leaving it for at least 20 minutes and uploading those logs will help, if we are not seeing consistent wake ups and sends in the logs every 2 and 10 minutes we have a different issue to influx upload.

Also in one of your early comments you suggest that you have unplugged it to run, but all logs suggest it is still on USB power. Are you running battery power and this is further weirdness or did I misunderstand that all testing is occurring on battery. Either battery or USB should work, but it might be worth seeing if you get different behaviour on battery power via the JST connector.

Finally, is the board behaving properly where the white light pulses for a few seconds each wake up period and then turns off (will pulse longer for the upload wake up as it finds and connects to wifi). A flashing or solid red light would indicate an issue (I don't think you will have this as not in the log), but if the white light stays on and stops pulsing, then I have seen that on the board and it will need a reset to clear. The fix is some new code to better manage the wifi disconnect, but only worth trying to mess with that if the board is not shutting down properly (white light fully goes of and poking the button causes it to wake up and take a reading fine).