pimoroni/enviro #200: Continuous red blink on enviro weather

NotSqrt, 2023-09-09:
Hello,

My enviro weather was at version 0.0.9, is now continuously blinking red.
Same thing on battery or USB power.
Upgrade to 0.0.10 or downgrade to older version does not change anything.
Even the flash_nuke.uf2 does not seem to change anything to the blinking red led.
The provisioning mode does no longer get triggered by poke + reset.

I manage to access the RPI-RP2 folder but there's only INDEX.HTM and INFO_UF2.TXT.
I've never been able to find the `config.py` or `log.txt` files, or make Thonny connect to the board.
Perhaps I'm missing something on my Ubuntu 22 LTS laptop ..

Any ideas ?
thanks !

sjefferson99, 2023-09-09:
The Pico is a little unusual in that the USB flash drive only ever presents those two files, even after you copy a UF2 to it, that doesn't show up. The filesystem presents differently and Thonny is the ideal way to view and manipulate those files easily.
For enviro, the red led is driven driven by the offboard RTC, s it's possible for the Pico to be in a good state if you've reflashed it and the red LED still flash.

I recommend flash nuking the board and then copying either the enviro UF2 or even a more vanilla pico or pimoroni picow UF2 to it and confirming you can now connect in Thonny by pressing the red stop button to force it to reconnect. If you then flash the board with the enviro v0.0.10 uf2 you should still be able to connect with Thonny and the provisioning mode should start once you disconnect thonny and reset so it runs main.py Once you reprovision the red light should clear as the code reconfigures the RTC.

NotSqrt, 2023-09-10:
Well, for some reason, 24 hours without power seems to have restored proper behaviour : no more red LED, provisioning is working, /dev/ttyACM0 is available, files like `config.py`, `log.txt` can now be seen in thonny ...

Thanks a lot !

closed 2023-09-10 by NotSqrt