pimoroni/enviro #201: No adafruit.io key

vaelen, 2023-09-14:
Just set up my board for the first time. Using v0.0.10. When it tries to log to adafruit.io the key is blank.

```
2023-09-14 09:41:55 [debug    / 105kB]   - upload issue 'not found - There is no feed with the key ''. You can create one using the feed creation API https://io.adafruit.com/api/docs/#create-feed or at https://io.adafruit.com/andrewyoung/feeds' (404 - Not Found)
```

When I connect the pico to USB normally it doesn't present a drive to get logs from, etc., and when I connect the pico with bootsel held down, I only get the files explaining how the uf2 uploader works, so I can't edit the config.py manually.

```
andrew@raspberrypi:/media/andrew/RPI-RP2 $ ls -la
total 28
drwxr-xr-x  2 andrew andrew 16384 Jan  1  1970 .
drwxr-x---+ 3 root   root    4096 Sep 14 18:50 ..
-r--r--r--  1 andrew andrew   241 Sep  6  2008 INDEX.HTM
-r--r--r--  1 andrew andrew    62 Sep  6  2008 INFO_UF2.TXT
```

Any ideas on what I might be doing wrong?

helgibbons, 2023-09-14:
Sounds like something went wrong during the provisioning process - there's instructions on how to connect to the board using Thonny to check the contents of config.py here: https://learn.pimoroni.com/article/getting-started-with-enviro#troubleshooting

Note that you only need to hold down the BOOTSEL button to get into DFU mode when flashing new firmware to the device, it won't show up as a drive at other times.

vaelen, 2023-09-14:
Ah, I didn't realize you have to use micropython to view the fs.

Here is the config.py file, passwords redacted. 
```
# enviro config file

# you may edit this file by hand but if you enter provisioning mode
# then the file will be automatically overwritten with new details

provisioned = True

# enter a nickname for this board
nickname = 'upstairs-herbs'

# network access details
wifi_ssid = '<redacted>'
wifi_password = '<redacted>'

# how often to wake up and take a reading (in minutes)
reading_frequency = 15

# how often to trigger a resync of the onboard RTC (in hours)
resync_frequency = 168

# where to upload to ("http", "mqtt", "adafruit_io", "influxdb")
destination = 'adafruit_io'

# how often to upload data (number of cached readings)
upload_frequency = 5

# web hook settings
custom_http_url = ''
custom_http_username = ''
custom_http_password = ''

# mqtt broker settings
mqtt_broker_address = ''
mqtt_broker_username = ''
mqtt_broker_password = ''
# mqtt broker if using local SSL
mqtt_broker_ca_file = None

# adafruit ui settings
adafruit_io_username = 'andrewyoung'
adafruit_io_key = '<redacted>'

# influxdb settings
influxdb_org = ''
influxdb_url = ''
influxdb_token = ''
influxdb_bucket = ''

# grow specific settings
auto_water = False
moisture_target_a = 50
moisture_target_b = 50
moisture_target_c = 50

# compensate for usb power
usb_power_temperature_offset = 4.5
```

vaelen, 2023-09-15:
I found the issue. You need to create a group in Adafruit.io called "enviro". Once you do that, the board updates without a problem.

https://github.com/pimoroni/enviro/blob/main/enviro/destinations/adafruit_io.py#L28

closed 2023-09-15 by vaelen