pimoroni/enviro #202: Enviro Weather - Is wifi_ssid limited as number of characters?

Fabri54, 2023-09-14:
Hi all,
I'm facing a weird behavior when configuring te Enviro Weather. At home, due to the layout of the flat, to the type of walls, wiring and plumbing, etc., I'm forced to use two access points.
With one of them the Enviro Weather provisioning works like a charm, with the other the connection to the WLAN hangs without ever getting connected.
Differences between the two WLANs:
The first one, working fine, has the wifi_ssid  6 characters long and the wifi_password  is 9 characters (numbers, letters and special characters)
The second one, with problems, has a 21 characters long wifi_ssid and a 15 characters long wifi_password. These latter, ssid and pswd contains just letters, the minus sign and numbers.
I've also tried editing manually the config.py file in both cases and flashed the latest FW. All this to no results. I'm therefore wondering if, somewhere, there is a limit to the length of the SSID name.
Many thanks to anybody willing to help me.
P.S. I cannot change the SSIDs as I've plenty of domotic devices connected and changing the name of the SSID would be an extremely long and cumbersome task.

Gadgetoid, 2023-09-14:
The maximum SSID length is 32 characters insofar as I'm aware, any limitation would be part of the WiFi driver and/or MicroPython rather than Enviro but - just in case - have you checked your config file to observe the full SSID there?

Any other differences between the networks?

Fabri54, 2023-09-14:
Apparently, in the config.py, the SSID is correctly captured at provisioning time, likewise the password.
The two AP are different, brand, and one is the master where the FTTH is connected and the other is working as an extender  this latter is the one that works without a glitch. The master is the one having problems with the Enviro Weather.

sjefferson99, 2023-09-14:
Is there anything of use in the log files comparing each AP connection attempt?

Fabri54, 2023-09-14:
Content of the config.py file after deleting the fie and resetting the Pico W

2000-01-01 00:52:16 [debug    / 135kB]   - running Enviro 0.0.9, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2000-01-01 00:52:16 [info     / 126kB]   - wake reason: usb_powered
2000-01-01 00:52:16 [debug    / 124kB]   - turn on activity led
2000-01-01 00:52:16 [info     / 122kB] > clock not set, synchronise from ntp server
2000-01-01 00:52:16 [info     / 120kB] > connecting to wifi network 'Vodafone-34350430-2p4'
2000-01-01 00:52:17 [error    / 115kB] ! failed to connect to wireless network Vodafone-34350430-2p4
2000-01-01 00:52:17 [error    / 113kB] ! failed to synchronise clock
2000-01-01 00:52:17 [info     / 111kB] > going to sleep
2000-01-01 00:52:17 [debug    / 109kB]   - clearing and disabling previous alarm
2000-01-01 00:52:17 [info     / 133kB]   - setting alarm to wake at 00:55am
2000-01-01 00:52:17 [info     / 131kB]   - shutting down
2000-01-01 00:52:17 [debug    / 129kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead

Gadgetoid, 2023-09-14:
Try tweaking these lines:

https://github.com/pimoroni/enviro/blob/b5bff82f9001409c9683be43b9f2114e7da8b430/enviro/__init__.py#L179-L182

Add ` wlan.config(pm=0xa11140)` right below `wlan.active(True)`, so it looks like:

```python
 rp2.country("GB")
 wlan = network.WLAN(network.STA_IF)
 wlan.active(True)
 wlan.config(pm=0xa11140)
 wlan.connect(wifi_ssid, wifi_password)
 ```
 
 (country should probably also be in the config file here :grimacing:)

sjefferson99, 2023-09-14:
You could try the wifi improvements in PR #199, it improves reliability, abstracts country code to config and also improves logging.

Fabri54, 2023-09-14:
Last update for today. 
I must have royally screwed up something editing __init__.py as the Enviro Weather doesn't even try to connect to WiFi, neither to the previously OK WLAN, nor to the one that was not OK.
Hem...I've to say that, not only I added the wlan.config(pm0xa11140) line but also changed the rp2.country("GB") line to  rp2.country("IT"), being in Italy.
At first seeing that WiFi connection didn't even start I  thought to be stumbled into a "Brexit favorable " HW+FW combination. However, even restoring everything to the initial configuration now the board limits itself to take a measurement save it as a .txt file and nothing else.
To recap, currently the Enviro Weather doesn't connect to WiFi whatever is the WLAN available. Likely I will flash once more the FW and start from scratch tomorrow.
P.S. I haven't understood what to do  with improvement in PR #199. It's above my capabilities.

sjefferson99, 2023-09-14:
@Fabri54 In this case, after reflashing to the latest enviro firmware, you can get away with deleting the contents of the enviro/\_\_init\_\_.py file and replacing it with the contents of the version linked below in Thonny (Note copy paste text not copy paste file as Thonny doesn't like folder structures in my experience).

https://raw.githubusercontent.com/pimoroni/enviro/75b403068333b427fb6b8374d2642ab04c91f2d8/enviro/__init__.py

Then add the below line into config.py, correcting for your country as appropriate
```
wifi_country = "GB"
```

Fabri54, 2023-09-15:
My apologies, I hate to make you waste your time because of my lack of experience, but the problem is still there.

 I nuked the flash and reflashed the FW. I went through provisioning with the WLAN that I knew to be OK and the Enviro connected and started to take measurements. I replaced the --init--.py with the one you pointed me to and also added the wifi_country = "IT" in the config.py, just above the wifi-ssid and wifi_password lines. 
Manually modified the wifi_ssid and wifi_password o the name/password of the WLAN that had problems. 
I reset the Enviro but the problem is still there. Here is an excerpt of the log file

2000-01-01 00:00:04 [info     / 135kB] > performing startup
2000-01-01 00:00:04 [debug    / 132kB]   - running Enviro 0.0.9, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2000-01-01 00:00:04 [info     / 123kB]   - wake reason: usb_powered
2000-01-01 00:00:04 [debug    / 121kB]   - turn on activity led
2000-01-01 00:00:04 [info     / 119kB] > clock not set, synchronise from ntp server
2000-01-01 00:00:04 [info     / 117kB] > connecting to wifi network 'Vodafone-34350430-2p4'
2000-01-01 00:00:04 [info     / 113kB] > MAC: 28:cd:c1:08:0a:17
2000-01-01 00:00:05 [info     / 111kB] > active: 0, status: 0 (Link is down)
2000-01-01 00:00:05 [info     / 109kB] > Ready for connection!
2000-01-01 00:00:05 [info     / 132kB] > Connecting to SSID Vodafone-34350430-2p4 (password: **************)...
2000-01-01 00:00:05 [info     / 128kB] > active: 1, status: -2 (No matching SSID found (could be out of range, or down))
2000-01-01 00:00:05 [error    / 125kB] ! Failed to connect to SSID Vodafone-34350430-2p4 (password: i2jukebcpz4m7j2): No matching SSID found (could be out of range, or down)
2000-01-01 00:00:05 [error    / 123kB] ! failed to synchronise clock
2000-01-01 00:00:06 [info     / 121kB] > going to sleep
2000-01-01 00:00:06 [debug    / 119kB]   - clearing and disabling previous alarm
2000-01-01 00:00:06 [info     / 117kB]   - setting alarm to wake at 00:05am
2000-01-01 00:00:06 [info     / 114kB]   - shutting down
2000-01-01 00:00:06 [debug    / 112kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead

I'm 100% sure that the SSID is correct. At this point in time I'm a bit perplexed. I might solve the problem by moving to the room with thee other WLAN. Currently I don't plan to use the Enviro here but in the country house and want to add a web server, create a dedicated WLAN, based on Raspberry Pi, the sensors and have a display showing the collected data.
However, I'd really like to understand the root cause of the problem even if, apparently, I may circumvent it adopting 6-7 characters long SSID.
Many thanks for the help and for your time

Fabri54, 2023-09-15:
Thank you! The SSID can be read in the excerpt of the log.txt file I've added to my last post and it is correct. During provisioning is correctly captured and I also tried typing it into the config.py file. The AP does offer 2.4GHz and 5GHz but the SSIDs are different (all IoT / domotic devices are connected to 2.4GHz, the others, iPad, iPhone Smart TV, etc. to 5GHz).
I'll google the status error. In any case I'm just coming from the room where, connected to the other WLAN, the Enviro Weather is quietly and stably taking measurements since half an hour. It is connected to a LiPo battery as, in my mind, it will be powered by a solar panel which will also charge the LiPo during sleep times. I'll post my findings in the coming days.
Many thanks again

Fabri54, 2023-09-16:
This is weird indeed. This morning I nuked the flash and installed on the Enviro Weather just MicroPython v1.20.0  (latest stable on 2023-04-26)
Then I loaded a very simple program which connects the Pico W to the WLAN (the infamous one) and queried google.com to get a time stamp. 
It worked without any glitch. This drives me to think that the problem comes from the FW provided for the Enviro and not from the AP / Pico.
Any idea?

sjefferson99, 2023-09-16:
That is odd and does indeed swing suspicion back to Enviro. I can't think of anything off hand that Enviro would do that could interfere with that. The wifi improvement code PR was based on the rpi manual on how to connect, so in theory it should be very similar.

I'm sorry I'm not sure which program you might be using (infamous or not, I am good at avoiding famousness in most places in life, sometimes surprisingly so), assuming it is open source, can you post the code or a link to git etc. here so we can compare?

Fabri54, 2023-09-16:
Sorry, with infamous WLAN I intended to refer to the home WLAN (SSID Vodafone-34..........) which doesn't connect to the Enviro Weather, which is running Enviro FW,  but that connects  to it when the Pico PI has just MicroPython on it and runs a simple WiFi program.
Actually, even if I'm curious to understand why there are these two different behaviors, I'm considering to drop this investigation and to cannibalize just the Enviro Weather fw ( main.py, __init__.py and related programs) for what concerns the reading of the sensors, BME280, Wind Vane and Anemometer.
My initial project was, and still is to have an external barometer with wind measurement, a few internal sensors (temp and humidity) each running a web server, all hooked to a WLAN, not necessarily always connected to Internet, and, in house, a Raspberry Pi with a 7" touchscreen which runs as an electronic frame with family pictures (to appease wife) and, at a touch, displays all available weather info for a certain amount of seconds.
For all this I believe I don't need the whole set of features provided by the Enviro FW.
P.S. The squeamish AP, provided by Vodafone, is based on a Huawei HW and has its own proprietary FW

Fabri54, 2023-09-17:
Last update for today : Nuked and flashed again Enviro FW. Deleted all programs and left just the Enviro directory structure empty.
Main.py now is just a simple program to connect to the WiFi and prints the number of seconds taken or the failure to do so.
It works with both home WiFi and takes 3 sec to connect to each of them. Next step to add a loop to connect to the strongest WiFi signal (not sure it is possibile. On ESP32, WiFi library (C++) allows to do so. Intent is to avoid Enviro  provisioning and have a list of WiFi SSID/passwords in a dedicated file to be used but connect loop in main.py or __init__.py, if it makes sense. Being a beginner I'm open to suggestions.
Guess I should then close this post

Fabri54, 2023-09-20:
This to close the thread. Many thanks to all those which took their time to help me

closed 2023-09-20 by Fabri54