pimoroni/enviro #206: remote force clock sync

adamczarnecki, 2023-12-12:
is there any way to remotely force device to sync rtc, or feed current time by http response? or can you guide me how to do that?

sjefferson99, 2023-12-12:
Once provisioning is complete, there is no listening interface on the board you could remotely connect to, it simply fires data out at configured intervals.

The rp2040 chip syncs it's time to the onboard RTC every startup.

NTP is set on startup if the date is not set or obviously wrong, however, there is a config option "resync_frequency" which will trigger an NTP resync after this many hours even if the RTC appears to be a valid time (not many years in the past). This is designed to keep clock drift to a minimum.

What issue are you experiencing that is making you want to initiate an NTP sync remotely?

Chattox, 2024-03-05:
Not OP but having a potentially similar issue; my enviro weather's time seems to get 1 hour out of sync relatively regularly. Restarting the board fixes it, but it ends up an hour behind within a day or so.

sjefferson99, 2024-03-05:
You can check the log for successful time sync entries and also see if you are seeing a constant large drift (more than a few seconds per day) or if it's a sudden jump, might need modification of the time sync code to log when it checks and what the difference is to write out to a new file copying the reading storage or logging function.

I seem to recall somewhere that some people had issues with the NTP server URL being pool.ntp.org and adding the UK prefix helped, maybe this might help here? I think that URL config is within the phew folder somewhere from memory.

A hack might be to aggressively reduce the resync_frequency in config to do requery NTP more regularly, assuming it is successfully doing so.

Chattox, 2024-03-06:
Yeah I was thinking of reducing the resync frequency to like an hour or so because mine is currently out of sync by ~4 hours and it's only been about 3 days since I last reset the board. I imagine having it resync the clock much more frequently would do a number on battery life?

sjefferson99, 2024-03-06:
Resyncing the clock would mean connecting to wifi, If you're connecting to wifi to send results very infrequently say every few hours then adding an extra connection for an NTP sync in every few hours would impact that if they didn't line up. If you're pushing results every 15 minutes or more frequently then an extra connection to wifi is less of a noticeable uplift. I am not sure how much power sending an extra few packets adds, certainly some, but not nearly as much as having to look for and connect to wifi and negotiate that connection to send any kind of data. So the relative answer depends on your current data push schedule.

That said, increasing the resync frequency to shorter than every few days is a short term hack and would be masking an underlying issue. The onboard RTC should not be drifting that fast and the pico syncs to it on startup.

adamjuraczarnecki, 2024-03-06:
i have network of 10+ enviro urban and from my experience issue appears when rasberry pico or whole device crush and bricks with enough power to keep last time in memory. when for example baterry voltage is to low to send reading via wifi but enough to hold last time in memory. When after change of batteries, device check last known time, this time is less than time needed to force new sync, so device is going to run with time minus time it was bricked. And how to explain to non technical gu what to to to reset this clock :D

Chattox, 2024-03-09:
@adamjuraczarnecki I hadn't even thought of that! The drift started on mine soon after I replaced the batteries, so I've wiped the time sync files and forced a resync; here's hoping that was the issue, I'll know within the next day or so 😁

Chattox, 2024-03-23:
So as a little update this absolutely did not fix my issue and it reoccurred again within a couple of days. Having done some googling, it seems that the 2 x AAA battery pack shipped with the enviro isn't enough to power the enviro weather station so I suspect this may be the cause of my issue as I found [this post](https://forums.pimoroni.com/t/envirogrow-solid-red-led-and-doesnt-boot-when-using-jst-ph-power-connector/20101/6) from Pimoroni's forums that seems to agree. Gonna order a 3 x AAA battery pack and see if that helps :D