pimoroni/enviro #207: Errors

GrahamDyer, 2023-12-29:
Been trying to run a weather station for some time now but am getting the following errors. 
> connecting to wifi network 'MyNetwork'
! failed to connect to wireless network MyNetwork
2023-12-29 18:43:07 [error    / 132kB]   - cannot upload readings, wifi connection failed
2023-12-29 18:43:07 [error    / 129kB] ! reading upload failed
My WIFI network is working perfectly and all my other devices connect 100%. SSID and password have been checked to be correct many times.
This from earlier in the log:
> connecting to wifi network 'MyNetwork'
2023-12-29 18:15:06 [info     / 134kB]   - ip address:  192.168.0.44
2023-12-29 18:15:07 [info     / 127kB] > uploading cached readings to Adafruit.io: XYZ
So then it connected fine but it will not later?

The other error is 
> uploading cached readings to Adafruit.io: XYZ
2023-12-29 18:42:56 [info     / 112kB]   - uploaded 2023-12-29T18_26_33Z.json
2023-12-29 18:42:57 [info     /  91kB]   - uploaded 2023-12-29T18_30_05Z.json
2023-12-29 18:42:58 [info     / 114kB]   - uploaded 2023-12-29T18_32_04Z.json
2023-12-29 18:42:59 [info     /  98kB]   - **cannot upload '2023-12-29T18_32_46Z.json' - rate limited**
2023-12-29 18:42:59 [info     /  95kB] > going to sleep for 1 minute(s)
What is "rate limited" and why would I get that error? reading_frequency = 15 and upload_frequency = 3

GrahamDyer, 2024-01-02:
On router A (Tenda 442720) I get this:
`
2024-01-02 12:16:52 [info / 121kB] > connecting to wifi network 'MyNetwork'`

On router B (ZTE 4G MP286C) I get this:

```
2024-01-02 12:30:07 [error    / 119kB] ! failed to connect to wireless network MyNetwork
2024-01-02 12:30:07 [error    / 116kB]   - cannot upload readings, wifi connection failed
2024-01-02 12:30:07 [error    / 114kB] ! reading upload failed

```
Using latest v0.0.10 release.

Could this be a Raspberry Pi Zero W problem?

sjefferson99, 2024-01-03:
The rate limiting is a limit to how many submissions you can make to the adafruit API per hour. For the free version I think it's something small like 10. With the setting you have I suspect it's from repeatedly resetting the board to try and troubleshoot the network issues. You should find this error goes away once left to settle down assuming you keep the uploads to fewer than the free tier rate limit.

The latest release does not incorporate a number of improvement and fixes that are still stuck as pull requests. One of these addresses a number of networking issues and may fix yours. If you can run the code in PR #199 https://github.com/pimoroni/enviro/pull/199 you might find the network behaves better.