pimoroni/enviro #209: Enviro Weather Red Light v0.0.10

JohnRSim, 2024-02-03:
Just received Enviro Weather ran through the update steps getting the red flashing light.
I looked at the error logs would this be causing the error light to flash?

2024-02-03 14:26:56 [error    /  89kB]   - failed to fetch time from ntp server
2024-02-03 14:26:56 [error    / 133kB] ! failed to synchronise clock

I ran through the troubleshooting guide but still getting this issue on both USB and powerpack.
Can see the device logging into the router - don't think it's an issue connecting to the wifi.

I'll have a look at the config - is there anything else I should check?
Thanks

```
Moved Permanently) [241ms]
2021-01-01 00:01:07 [info     / 103kB] > GET /wrong-host-redirect (200 OK) [151ms]
2021-01-01 00:01:08 [info     / 110kB] > POST /provision-step-1-nickname (301 Moved Permanently) [666ms]
2021-01-01 00:01:08 [debug    /  81kB] > parsed template: enviro/html/header.html (took 22 ms)
2021-01-01 00:01:08 [debug    /  95kB] > parsed template: enviro/html/header-mini.html (took 14 ms)
2021-01-01 00:01:08 [debug    /  83kB] > parsed template: enviro/html/footer.html (took 14 ms)
2021-01-01 00:01:08 [debug    /  79kB] > parsed template: enviro/html/provision-step-2-wifi.html (took 313 ms)
2021-01-01 00:01:09 [info     /  76kB] > GET /provision-step-2-wifi (200 OK) [591ms]
2021-01-01 00:01:09 [info     / 126kB] > GET /hotspot-detect.html (301 Moved Permanently) [390ms]
2021-01-01 00:01:09 [info     / 107kB] > GET /networks.json (200 OK) [878ms]
2021-01-01 00:01:10 [info     /  96kB] > GET /images/logo-weather.svg (200 OK) [986ms]
2021-01-01 00:01:10 [info     /  91kB] > GET /wrong-host-redirect (200 OK) [927ms]
2021-01-01 00:01:11 [info     / 106kB] > GET /images/pirate-logo.svg (200 OK) [2539ms]
2021-01-01 00:01:11 [info     / 102kB] > GET /images/guide-qr-code.svg (200 OK) [2583ms]
2021-01-01 00:01:27 [info     / 104kB] > POST /provision-step-2-wifi (301 Moved Permanently) [735ms]
2021-01-01 00:01:27 [debug    / 100kB] > parsed template: enviro/html/header.html (took 21 ms)
2021-01-01 00:01:28 [debug    /  94kB] > parsed template: enviro/html/header-mini.html (took 13 ms)
2021-01-01 00:01:28 [debug    /  79kB] > parsed template: enviro/html/footer.html (took 12 ms)
2021-01-01 00:01:28 [debug    /  77kB] > parsed template: enviro/html/provision-step-3-logging.html (took 324 ms)
2021-01-01 00:01:28 [info     /  74kB] > GET /provision-step-3-logging (200 OK) [494ms]
2021-01-01 00:01:28 [info     / 125kB] > GET /hotspot-detect.html (301 Moved Permanently) [348ms]
2021-01-01 00:01:28 [info     / 118kB] > GET /images/pirate-logo.svg (200 OK) [615ms]
2021-01-01 00:01:29 [info     / 111kB] > GET /images/logo-weather.svg (200 OK) [751ms]
2021-01-01 00:01:29 [info     / 107kB] > GET /wrong-host-redirect (200 OK) [702ms]
2021-01-01 00:01:29 [info     / 104kB] > GET /images/guide-qr-code.svg (200 OK) [881ms]
2021-01-01 00:01:29 [info     /  97kB] > GET /hotspot-detect.html (301 Moved Permanently) [200ms]
2021-01-01 00:01:29 [info     /  91kB] > GET /wrong-host-redirect (200 OK) [195ms]
2021-01-01 00:01:32 [info     / 101kB] > POST /provision-step-3-logging (301 Moved Permanently) [648ms]
2021-01-01 00:01:32 [debug    / 100kB] > parsed template: enviro/html/header.html (took 35 ms)
2021-01-01 00:01:33 [debug    /  94kB] > parsed template: enviro/html/header-mini.html (took 12 ms)
2021-01-01 00:01:33 [info     / 112kB] > GET /hotspot-detect.html (301 Moved Permanently) [179ms]
2021-01-01 00:01:33 [debug    / 100kB] > parsed template: enviro/html/footer.html (took 9 ms)
2021-01-01 00:01:33 [debug    /  97kB] > parsed template: enviro/html/provision-step-4-destination.html (took 514 ms)
2021-01-01 00:01:33 [info     /  95kB] > GET /provision-step-4-destination (200 OK) [739ms]
2021-01-01 00:01:33 [info     /  94kB] > GET /wrong-host-redirect (200 OK) [472ms]
2021-01-01 00:01:33 [info     /  81kB] > GET /images/logo-weather.svg (200 OK) [635ms]
2021-01-01 00:01:34 [info     / 111kB] > GET /images/pirate-logo.svg (200 OK) [495ms]
2021-01-01 00:01:34 [info     / 104kB] > GET /images/guide-qr-code.svg (200 OK) [573ms]
2021-01-01 00:01:34 [info     /  98kB] > GET /hotspot-detect.html (301 Moved Permanently) [209ms]
2021-01-01 00:01:34 [info     /  92kB] > GET /wrong-host-redirect (200 OK) [189ms]
2021-01-01 00:02:01 [info     / 107kB] > POST /provision-step-4-destination (301 Moved Permanently) [1218ms]
2021-01-01 00:02:02 [debug    / 104kB] > parsed template: enviro/html/header.html (took 26 ms)
2021-01-01 00:02:02 [debug    /  99kB] > parsed template: enviro/html/header-mini.html (took 14 ms)
2021-01-01 00:02:02 [debug    /  90kB] > parsed template: enviro/html/footer.html (took 9 ms)
2021-01-01 00:02:02 [debug    / 118kB] > parsed template: enviro/html/provision-step-5-done.html (took 352 ms)
2021-01-01 00:02:02 [info     / 115kB] > GET /provision-step-5-done (200 OK) [889ms]
2021-01-01 00:02:02 [info     / 107kB] > GET /hotspot-detect.html (301 Moved Permanently) [392ms]
2021-01-01 00:02:03 [info     / 109kB] > GET /images/pirate-logo.svg (200 OK) [520ms]
2021-01-01 00:02:03 [info     / 105kB] > GET /images/logo-weather.svg (200 OK) [638ms]
2021-01-01 00:02:03 [info     / 100kB] > GET /wrong-host-redirect (200 OK) [556ms]
2021-01-01 00:02:03 [info     /  97kB] > GET /images/guide-qr-code.svg (200 OK) [750ms]
2021-01-01 00:02:03 [info     /  90kB] > GET /hotspot-detect.html (301 Moved Permanently) [80ms]
2021-01-01 00:02:03 [info     /  84kB] > GET /wrong-host-redirect (200 OK) [206ms]
2024-02-03 14:26:40 [info     / 138kB] > performing startup
2024-02-03 14:26:40 [debug    / 135kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-02-03 14:26:40 [info     / 125kB]   - wake reason: usb_powered
2024-02-03 14:26:40 [debug    / 123kB]   - turn on activity led
[... 20 lines cut ...]
2024-02-03 14:32:14 [info     / 138kB] > performing startup
2024-02-03 14:32:14 [debug    / 135kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-02-03 14:32:14 [info     / 125kB]   - wake reason: usb_powered
2024-02-03 14:32:14 [debug    / 123kB]   - turn on activity led
2024-02-03 14:32:14 [info     / 121kB] > clock not set, synchronise from ntp server
2024-02-03 14:32:14 [info     / 119kB] > connecting to wifi network 'FenOakStar2.4'
2024-02-03 14:32:19 [info     / 127kB]   - ip address:  192.168.50.14
2024-02-03 14:32:30 [error    / 121kB]   - failed to fetch time from ntp server
2024-02-03 14:32:30 [error    / 119kB] ! failed to synchronise clock
2024-02-03 14:32:30 [info     / 117kB] > going to sleep
2024-02-03 14:32:30 [debug    / 115kB]   - clearing and disabling previous alarm
2024-02-03 14:32:30 [info     / 112kB]   - setting alarm to wake at 14:45pm
2024-02-03 14:32:30 [info     / 110kB]   - shutting down
2024-02-03 14:32:30 [debug    / 108kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2024-02-03 14:33:26 [debug    / 135kB]   - reset
2021-01-01 00:00:06 [info     / 128kB] > entering provisioning mode
2021-01-01 00:00:06 [info     / 133kB] > auto detecting board type
2021-01-01 00:00:06 [info     / 131kB]   - weather
2021-01-01 00:00:06 [info     / 129kB] > going into access point mode
2021-01-01 00:00:07 [info     / 127kB]   - 192.168.4.1
2021-01-01 00:00:07 [info     / 125kB] > starting dns server...
2021-01-01 00:00:07 [info     / 123kB] > starting catch all dns server on port 53
2021-01-01 00:00:07 [info     / 121kB] > creating web server...
2021-01-01 00:00:07 [info     / 116kB] > waiting for a client to connect
```

JohnRSim, 2024-02-03:
```
# enviro config file

# you may edit this file by hand but if you enter provisioning mode
# then the file will be automatically overwritten with new details

provisioned = True

# enter a nickname for this board
nickname = 'fenoak-weather'

# network access details
wifi_ssid = 'FenOakStar2.4'
wifi_password = 'PimoroniTeamAreTheBest'

# how often to wake up and take a reading (in minutes)
reading_frequency = 15

# how often to trigger a resync of the onboard RTC (in hours)
resync_frequency = 168

# where to upload to ("http", "mqtt", "adafruit_io", "influxdb")
destination = 'adafruit_io'

# how often to upload data (number of cached readings)
upload_frequency = 10

# web hook settings
custom_http_url = ''
custom_http_username = ''
custom_http_password = ''

# mqtt broker settings
mqtt_broker_address = ''
mqtt_broker_username = ''
mqtt_broker_password = ''
# mqtt broker if using local SSL
mqtt_broker_ca_file = None

# adafruit ui settings
adafruit_io_username = 'JRSim_UIX'
adafruit_io_key = 'PimoroniTeamAreTheBest'

# influxdb settings
influxdb_org = ''
influxdb_url = ''
influxdb_token = ''
influxdb_bucket = ''

# grow specific settings
auto_water = False
moisture_target_a = 50
moisture_target_b = 50
moisture_target_c = 50

# compensate for usb power
usb_power_temperature_offset = 4.5

```

sjefferson99, 2024-02-03:
The issue is indeed failure to sync with NTP server, the main code loop will bail and wait for the next wakeup to try again while flashing the red light if the NTP sync fails. This is reflected by the log entry "! failed to synchronise clock"

The cause of this is the NTP module fails to populate the timestamp variable with a valid time, which aligns to log output "  - failed to fetch time from NTP server".

Given you have a valid IP address logged above you have successfully connected to your AP and sent/received packets to get a DHCP response, so I would suggest this is a network issue between your AP/firewall/router preventing the Pico making the call to the NTP server rather than any wireless config you could adjust on the Pico.

Could you try setting an hotspot up on your phone and connecting to that instead of your main wireless, if that works then you can confirm it's that specific connection that has an issue.

JohnRSim, 2024-02-03:
I connected via mobile hotspot and that worked.
Then reset and connected back to AP (Asus GT-AX600) and I think it is now working.. just waiting and watch to see if it turns back on..

I have this connected via StarlinkISP maybe a temporary intermittent issue but didn't notice any connectivity issues?

JohnRSim, 2024-02-04:
Which NTP server is being used I'm getting red lights again flashing..

sjefferson99, 2024-02-04:
The failing function is calling this code (phew.ntp): https://github.com/pimoroni/phew/blob/b33a44bffeb1354b730b9572410232b10f6e62e9/phew/ntp.py
So it looks like pool.ntp.org.

daviestim, 2024-02-04:
I had this problem (or a very similar one). I worked around it by changing the NTP server (line 4)
`  ntp_host = "uk.pool.ntp.org"`
You could try picking one for your location and see if it works? https://www.ntppool.org/en/

closed 2024-02-06 by JohnRSim