pimoroni/enviro #210: Enviro Urban won't work, LED always on, It says usb powered when it is powered by batteries.

A1a01, 2024-02-11:
Hello my Enviro Urban works fine for a couple of hours and then it stops working, leaving the white LED turned on.
Checking the logs it says it was powered by usb (which is not, it is powered by a LiPo battery) and I think that's the issue with the board. Any solution to the error of usb powered?
The logs where it shows usb powered:
2024-02-10 22:48:18 [info     / 115kB]   - shutting down
2024-02-10 22:48:18 [debug    / 113kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2024-02-10 22:54:00 [debug    / 111kB]   - reset
2024-02-10 22:54:03 [info     / 138kB] > performing startup
2024-02-10 22:54:03 [debug    / 135kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-02-10 22:54:03 [info     / 115kB]   - wake reason: rtc_alarm
2024-02-10 22:54:03 [debug    / 113kB]   - turn on activity led
2024-02-10 22:54:03 [debug    / 108kB] > 99 blocks free out of 212
2024-02-10 22:54:03 [debug    / 106kB] > taking new reading
2024-02-10 22:54:03 [info     / 133kB]   - seconds since last reading: 360
2024-02-10 22:54:04 [debug    / 130kB]   - starting sensor
2024-02-10 22:54:04 [debug    / 128kB]   - wait 5 seconds for airflow
2024-02-10 22:54:09 [debug    / 109kB]   - taking pms5003i reading
2024-02-10 22:54:09 [debug    / 106kB]   - taking microphone reading
2024-02-10 22:54:16 [info     / 119kB] > going to sleep
2024-02-10 22:54:16 [debug    / 117kB]   - clearing and disabling previous alarm
2024-02-10 22:54:16 [info     / 115kB]   - setting alarm to wake at 23:00pm
2024-02-10 22:54:16 [info     / 113kB]   - shutting down
2024-02-10 22:54:16 [debug    / 110kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2000-01-01 00:00:04 [info     / 138kB] > performing startup
2000-01-01 00:00:04 [debug    / 135kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2000-01-01 00:00:04 [info     / 129kB]   - wake reason: usb_powered
2000-01-01 00:00:04 [debug    / 127kB]   - turn on activity led
2000-01-01 00:00:04 [info     / 125kB] > clock not set, synchronise from ntp server

sjefferson99, 2024-02-11:
The USB sense is done in enviro/__init_.py It simply reads wlgpio2 (which is connected to vbus) and if there is a voltage then that means a USB cable is supplying voltage. The only other way I know that it would report USB connected is if somehow you have connected the lipo or any other positive rails to vbus (pin40).

I have also seen 100% consistent behaviour when reporting on USB power or not which supports the simplicity of the code.

Any chance you have something connected to vbus somehow?

[image]

A1a01, 2024-02-23:
I haven't connected anything to vbus and I tried implementing #144 and #199 and it seemed to work fine for a couple of weeks but today the white LED happened again. I added a photo to show how I found the board.
[image]
And also the log file doesn't show any indication to what seems to be the issue, 
2024-02-22 23:21:43 [info     / 108kB]   - uploaded 2024-02-22T23_21_35Z.json
2024-02-22 23:21:43 [info     / 106kB] > Disconnecting wireless after upload
2024-02-22 23:21:43 [info     / 104kB] > going to sleep
2024-02-22 23:21:43 [debug    / 102kB]   - clearing and disabling previous alarm
2024-02-22 23:21:43 [info     / 100kB]   - setting alarm to wake at 23:28pm
2024-02-22 23:21:43 [info     /  98kB]   - shutting down
2024-02-22 23:21:43 [debug    /  96kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
2000-01-01 00:00:05 [info     / 131kB] > performing startup
2000-01-01 00:00:05 [debug    / 129kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2000-01-01 00:00:05 [info     / 123kB]   - wake reason: usb_powered
2000-01-01 00:00:05 [info     / 121kB]   - setting default alarm to wake at 00:03am
2000-01-01 00:00:05 [debug    / 118kB] > delayoff set on gpio2 for 120000 ms

sjefferson99, 2024-02-24:
I've had a look at the schematics for the enviro and the pico-w and you should only get a digital 1 on USB if there is significant voltage (quite near to 5v if I've understood the voltage divider on wl_gpio2 and assuming a standard threshold for the pin detecting a 1) on vbus (bottom left in your picture).

[Enviro schematic](https://cdn.shopify.com/s/files/1/0174/1800/files/enviro_weather_schematic.pdf?v=1664452062)
[Pico-W power system schematic](https://datasheets.raspberrypi.com/picow/pico-w-datasheet.pdf) Page 18

I have measured voltage between ground and vbus and vsys on my board under battery power and I get 4v at vsys (which is the battery output right now) as expected and 0.75v on vbus. I think this is a result of the mosfet connecting vbus and vsys and should be well below the digital 1 input hence my board not reporting on USB power.

If you can measure vbus when it's on battery power and see if you get ~0.75v or something nearer 5v, maybe somehow 3.3v is getting passed to it which could potentially register as a digital 1.

If you have a voltage on vbus on battery only then I think that might be a board fault, if not then that is very mysterious.