pimoroni/enviro #214: Rain and/or wind sensors causing Enviro Weather to die and clear RTC

Chattox, 2024-04-02:
Having a bit of a weird issue I'm struggling to properly diagnose. Whenever I have the enviro board hooked up to the wind and rain sensors supplied as part of the Enviro Weather, the board eventually crashes after a few hours and stops waking up to take/upload readings. There are no error/status lights showing, and the logs just show the last successful reading and nothing else, no errors or anything. Disconnecting the enviro and pressing the Poke or reset buttons will wake it and get it working again, and upon checking the logs the RTC has been completely cleared and requires resyncing each time.

I've no idea what's causing this, I initially thought maybe the 2 x AAA battery pack supplied wasn't giving enough voltage so upgraded to a 3 x AAA pack but this made no difference. It was fine for the first few weeks, but then this started happening more or less out of the blue. Currently using the 0.0.10 firmware if that helps. Also attaching the most recent log showing the dropping out and RTC clearing. It looks a bit like the rain sensor code got stuck in a loop or something at 15:06, but then it managed one more successful scheduled reading after that before dying so it's not like the loop itself killed the board. The batteries were fine too, after resetting the board it's been ticking along on my desk without the wind/rain sensors attached ever since on the same batteries.
[enviro_logs_29-03-24.txt](https://github.com/pimoroni/enviro/files/14837683/enviro_logs_29-03-24.txt)

CBDesignS, 2024-04-02:
good luck waiting on the maintainer helping you.
we were told in January that will get updates and fixes, then in February and again in March but as it goes we are still waiting and its now April. 

A newer patched firmware does exist with fixes by the users that may help you.
https://github.com/pimoroni/enviro/actions/runs/7545806918/artifacts/1174690874 (think this is the correct link )

Chattox, 2024-04-03:
Yeeaah, I had a poke around here and on the discord server before posting this and it seems like the ball has been dropped in terms of maintaining just about every product they sell 😅 It's no worries though, I'm hoping it's a code issue rather than a hardware one so the problem will be fixable by us users instead. I'll give that new firmware a try and see how it goes :)

MrDrem, 2024-04-03:
I've installed that firmware as of last night, but not connected the device back it up. Is there a link to the code that is being used for this version, as I'd love to try and get the battery back working.

Chattox, 2024-04-03:
@MrDrem The patched firmware mentioned earlier is from #199 :)

So having put my enviro back out and hooked up to the wind/rain sensors, it died pretty quickly and stopped uploading, so this new firmware unfortunately doesn't fix it. However, looking at the new logs I _think_ I may have narrowed down what's causing the problem somewhat. It looks like the code that handles the rain sensor tripping is firing multiple times for some reason as there absolutely was not enough rain at 10:21 UTC for it to trip 3 times in the same second. Then, it managed one scheduled reading after that and died again just like previously. However, this time before taking it down I hit the poke button just out of curiosity which is shown as happening at 10:42 UTC in the logs, but was actually around 13:10 UTC so the RTC completely stopped counting time just under 13 minutes after it's last scheduled reading.
This leads me to think that something about the rain sensor code is causing the board to hang during the sleep process, but I've no idea why it carried on for almost 13 minutes before doing so.

Here are the latest logs:

```
2024-04-03 10:15:03 [debug    / 116kB] > 101 blocks free out of 212
2024-04-03 10:15:03 [debug    / 114kB] > taking new reading
2024-04-03 10:15:03 [info     / 109kB]   - seconds since last reading: 170
2024-04-03 10:15:05 [debug    / 123kB] > caching reading for upload
2024-04-03 10:15:05 [info     / 118kB] > 1 cache file(s) not being uploaded. Waiting until there are 4 file(s)
2024-04-03 10:15:05 [info     / 116kB] > going to sleep
2024-04-03 10:15:05 [debug    / 113kB]   - clearing and disabling previous alarm
2024-04-03 10:15:05 [info     / 111kB]   - setting alarm to wake at 10:30am
2024-04-03 10:15:05 [info     / 109kB]   - shutting down
2024-04-03 10:21:31 [info     / 135kB] > performing startup
2024-04-03 10:21:31 [debug    / 132kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-04-03 10:21:31 [info     / 122kB] > add new rain trigger at 2024-04-03T10:21:31Z
2024-04-03 10:21:31 [debug    / 119kB]   - wake reason: trigger
2024-04-03 10:21:31 [info     / 117kB] > going to sleep
2024-04-03 10:21:31 [debug    / 115kB]   - clearing and disabling previous alarm
2024-04-03 10:21:31 [info     / 113kB]   - setting alarm to wake at 10:30am
2024-04-03 10:21:31 [info     / 111kB]   - shutting down
2024-04-03 10:21:31 [info     / 135kB] > performing startup
2024-04-03 10:21:31 [debug    / 132kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-04-03 10:21:31 [info     / 121kB] > add new rain trigger at 2024-04-03T10:21:31Z
2024-04-03 10:21:31 [debug    / 118kB]   - wake reason: trigger
2024-04-03 10:21:31 [info     / 116kB] > going to sleep
2024-04-03 10:21:31 [debug    / 114kB]   - clearing and disabling previous alarm
2024-04-03 10:21:31 [info     / 112kB]   - setting alarm to wake at 10:30am
2024-04-03 10:21:31 [info     / 110kB]   - shutting down
2024-04-03 10:21:31 [info     / 135kB] > performing startup
2024-04-03 10:21:31 [debug    / 132kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-04-03 10:21:31 [info     / 121kB] > add new rain trigger at 2024-04-03T10:21:31Z
2024-04-03 10:21:31 [debug    / 117kB]   - wake reason: trigger
2024-04-03 10:21:31 [info     / 116kB] > going to sleep
2024-04-03 10:21:31 [debug    / 114kB]   - clearing and disabling previous alarm
2024-04-03 10:21:31 [info     / 112kB]   - setting alarm to wake at 10:30am
2024-04-03 10:21:31 [info     / 110kB]   - shutting down
2024-04-03 10:30:03 [info     / 135kB] > performing startup
2024-04-03 10:30:03 [debug    / 132kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-04-03 10:30:03 [info     / 122kB]   - wake reason: rtc_alarm
2024-04-03 10:30:03 [debug    / 120kB]   - turn on activity led
2024-04-03 10:30:03 [debug    / 116kB] > 100 blocks free out of 212
2024-04-03 10:30:03 [debug    / 114kB] > taking new reading
2024-04-03 10:30:03 [info     / 109kB]   - seconds since last reading: 900
2024-04-03 10:30:04 [debug    / 120kB] > caching reading for upload
2024-04-03 10:30:05 [info     / 100kB] > 2 cache file(s) not being uploaded. Waiting until there are 4 file(s)
2024-04-03 10:30:05 [info     /  98kB] > going to sleep
2024-04-03 10:30:05 [debug    /  96kB]   - clearing and disabling previous alarm
2024-04-03 10:30:05 [info     /  94kB]   - setting alarm to wake at 10:45am
2024-04-03 10:30:05 [info     /  92kB]   - shutting down
2024-04-03 10:42:55 [info     / 135kB] > performing startup
2024-04-03 10:42:55 [debug    / 132kB]   - running Enviro 0.0.10, MicroPython 856e08b1931b88271816a2f60648f6ff332235b2, enviro v1.20.4 on 2023-08-04
2024-04-03 10:42:55 [info     / 122kB]   - wake reason: button
2024-04-03 10:42:55 [debug    / 121kB]   - turn on activity led
2024-04-03 10:42:55 [debug    / 116kB] > 101 blocks free out of 212
2024-04-03 10:42:55 [debug    / 114kB] > taking new reading
2024-04-03 10:42:55 [info     / 109kB]   - seconds since last reading: 772
2024-04-03 10:42:57 [debug    / 123kB] > caching reading for upload
2024-04-03 10:42:57 [info     / 118kB] > 3 cache file(s) not being uploaded. Waiting until there are 4 file(s)
2024-04-03 10:42:57 [info     / 115kB] > going to sleep
[... cut ...]
2024-04-03 13:11:07 [info     /  99kB]   - uploaded 2024-04-03T13_11_00Z.json
2024-04-03 13:11:07 [info     /  97kB] > Disconnecting wireless after upload
2024-04-03 13:11:07 [info     /  95kB] > going to sleep
2024-04-03 13:11:07 [debug    /  92kB]   - clearing and disabling previous alarm
2024-04-03 13:11:07 [info     /  90kB]   - setting alarm to wake at 13:15pm
2024-04-03 13:11:07 [info     / 131kB]   - shutting down
2024-04-03 13:11:07 [debug    / 129kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead
```

CBDesignS, 2024-04-03:
The last message we got seemed to point that there is only one coder (or it seems that way) that looks after all of the product code and since the launch of the Rpi 5 they are spending all of their time trying to get that working with their hardware and prob new products so to me (personally) I think these things are now dead or end of life and once remaining stock is sold out they will bomb the product and leave it to die.. I have never used my Enviro Weather as it would just crash time after time after time on the bench while testing it before deploying outside. 
I keep comming back more in hope that they might pull a finger out and get things working so I can change from a Fineoffset 1080 to this but its looking like I will just keep plugging along with Home Assistant and a 433mhz reciever to get my local weather data (at least this way it works 100% for my needs)..

Good luck getting your setups going..

CBDesignS, 2024-04-04:
@Chattox thats the reason why we are stuck. when you update the firmware I think you can pull the latest pico code. There is supposed to be updated code but the maintainer said they think there are breaking changes and thats why we need them to sort the problems out. I would love to get this going with the main reason to get the DB data to log windspeed etc. If you have the code knowledge you can use the source they provide and build your own version but without the help of the community (that now seems to be loosing interest just like the maintainer) we are lost. I really hope this project does get some love but after 8 months I am loosing faith.. There are a couple of git pull requests aimed at the battery problem but when I tried the code it was just the same result, hang and then crash. It could also be that hardware has been changed between board production runs that use slightly different hardware so what code works on one board might fail on another board due to hardware differences or tollerences. The permanent usb power code seemed to work for a few users but it also didn`t work for others like me.

MrDrem, 2024-04-04:
I've been reworking the wind code on my personal copy of it on here. I've based it on the code that is supposed to have better wifi.

I've certainly got the 16 wind directions working correctly, and it's no longer double checking the value.

I've started on trying to improve the wind speed code, but I need to double check that my maths is correct, as the values that I'm getting back seem to be really low.

sjefferson99, 2024-04-05:
@Chattox I'm with you on trying where we can to fix it up, whether for fun or utility. 

This doesn't change my view that personally I think it's a great shame there isn't the resource at Pimoroni to better support products they are selling as hardware and some kind of Beta or MVP firmware solution with critical bug fixes supported for longer. But given the choice of what we have or nothing at all, I'll take what we have and hope they find a model that works better for us in the future.

However it's definitely a challenge when working with custom hardware and complex drivers written in C and added to custom firmware when your main user base is presumably Micropython/Circuitpython conversant at best. Compiling a new custom micropython firmware based on the Pimoroni fork is challenging and as yet I've had no success at a booting UF2 file. I am not an expert coder in Python or C, so I just keep having a go at it and see where I get.

My own board is running flawlessly for me (bar my own introduced bug causing odd wind gust error spikes), but I have moved away from the original intent of a low powered device running on batteries for months entering a low power sleep state. This is firstly as it seems to have some issues and more importantly, that only hints at the weather and a continuously running 4 Hz wind sample and other sensors around once a minute gives more helpful information for me.

As such my board (in a difficult place to regularly mess with) isn't running code that I can check the issue above with and try and recreate. I did look at the code and your logs and the RTC seemingly stopping and it seems to rule out all the likely potential code problems in the sleep function. The schematic suggests the RTC is powered by the main power supply, so it's currently a mystery how you managed to get your logs above with what I can see. I'll let you know if I manage to gain any further insight to how all that hangs together and anything else you could try.

Chattox, 2024-04-06:
Thank you @sjefferson99 :D 

I'm happy to keep working on this thing though I'm primarily from a typescript background professionally with python as a hobby language unfortunately so while I can mostly understand the code in this repo, the C drivers are mostly beyond me 😅

You're right about the problem not lying in the sleep functionality I think, as given enough time it does occasionally wake up on it's own again which if it were crashing I assume that wouldn't happen which makes me wonder if it's maybe a hardware issue. It worked absolutely fine for about a month or so after I first set it up and only then did I start having these problems, so I'm wondering if maybe some errant rain or something managed to get past the stevenson screen and short something on the board perhaps?

It's definitely not the rain sensor as I was previously thinking though, I left it running outside yesterday and overnight with just the wind sensors connected and no rain sensor at all and unfortunately the bug occurred again. I don't have access to the logs right now, but you can see from this shot of the logs from my API it was fine until 21:30 last night, then went AWOL overnight until coming back online at 07:38 this morning with the RTC having paused overnight as well. The date and time in square brackets is when the API received the data, and the timestamp given is the timestamp contained in the reading object (in UTC hence the hour difference).

[image]

It's since been functioning fine after that one hiccup last night though admittedly for only a few hours at this point. It's not backlogged data either, as you can clearly see the jump in temperature/lux where it started taking readings in the morning once it was light out. On top of this, the upload time being at 38 minutes past the hour is very odd as usually it's on a quarter hour increment like 15 minutes past, 30 minutes past etc which makes me think the RTC ran for a few minutes after whatever caused the problem before stopping, giving it this offset.

I've no intention of giving up on fixing this issue, I know your board is running different code and so hard to test this stuff on your end but I greatly appreciate any help :D

sjefferson99, 2024-04-06:
@Chattox Shortly after I posted the above, my board stopped reporting updates, it didn't lose time, but needed a manual reset to restart it. I brought it back in and had it run on the bench and it lasted twice as long before and then stopped again. Although this time I could see in the logs that the second core was still running and reporting avg and gust winds (of 0 as it wasn't connected to the anemometer, so it wasn't a powerdown, which matches with most crashes reported on this board where often the white light is stuck on, which is driven by the pico and not the rtc like the red LED.

I don't have a lot of time for this in the immediate future, but when I can grab a spare hour I will see if I can selectively remove enviro code like the sleep function and run just mine and see if I can find which bit is introducing instability. Assuming that's possible, it's been a few months since I looked at my code, but will give it a go and report back anything that might help here.

Chattox, 2024-04-09:
So I left mine going over the weekend with just the wind sensors connected to see if maybe it was them somehow, and sure enough the board died repeatedly again. Then for completionism's sake I left it running outside from yesterday evening in the stevenson screen but with no extra sensors attached, so the configuration it's been running absolutely fine on my desk for. At about 20:15 it died and still hasn't resurfaced, so I guess that rules out it being any of the external sensors. 

At this point I'm a little out of ideas and am wondering if maybe it's just a faulty board or something, so I think I'm gonna get in touch with Pimoroni support and see about getting a replacement to test

Chattox, 2024-04-17:
So I got in touch with Pimoroni support and they sent me out a new enviro weather board which arrived yesterday. I set it up outside with the wind/rain sensors hooked up yesterday morning and so far it's been working without incident for just over 24 hours. It's also experiencing a _lot_ less wifi issues than the previous board did even before it started malfunctioning. The first board would fail to connect probably every other time it tried, while this board has failed exactly once just after provisioning and has been absolutely fine since.

I'm gonna leave it going for a few days and hopefully get some rain to be able to test that sensor too, but if everything goes fine I'll close this issue as it appears to have been a hardware malfunction :)

Chattox, 2024-04-20:
So the new board has been running for a few days now with 0 issue, not even a single failure to connect to wifi other than the one instance while setting it up, and it's recorded rainfall correctly too.

Gonna close this for now as it seems to have been a hardware fault rather than a code issue. Thanks everyone for your help and insights :)

closed 2024-04-20 by Chattox