pimoroni/enviro #216: Weather board fails to upload every other reading

MrBisquit, 2024-04-13:
My weather board is powered via the USB-Micro cable and when it uploads a reading and takes another one it seems to start flashing red and fails, but then a few tries later it succeeds (Probably a connection issue). I was wondering if there's anyway of me disabling it sleeping because I think that's what the issue is.

sjefferson99, 2024-04-23:
@MrBisquit First step would be to apply the wifi improvements code in pr #199 as that can sort all sorts of issues including ones similar to what you have described. The files for this are available in a recent build: https://github.com/pimoroni/enviro/actions/runs/8784437254

MrBisquit, 2024-05-01:
Alright, I'll try that next time I bring it in, thanks :)

(I'll close the issue once I've tested it and if it works, which may be a while depending on when I next bring it in)

sjefferson99, 2024-05-01:
@MrBisquit This code is now in main, so upgrade to v0.2.0 and see how you go.

MrBisquit, 2024-12-09:
I feel like there could be an option for like if it's going to be plugged into a constant power source (instead of batteries) to have the option to never disconnect from Wi-Fi so that it could possibly prevent this issue.