pimoroni/enviro #222: Enviro Grow failing when communicating with Home Assistant

MazenS, 2024-05-02:
Hello there - I'm at my wits end.
I'm trying to get my Enviro Grow to communicate with Home Assistant.
I'm starting to think there is something wrong with the Grow as opposed to my HA setup - though that could be the case.

MicroPython v1.22.2, enviro v1.22.2 on 2024-03-06; Raspberry Pi Pico W with RP2040

```
2021-01-01 00:00:02 [warning  / 135kB] > config setting 'mqtt_broker_ca_file' missing, please add it to config.py
2021-01-01 00:00:02 [warning  / 143kB] > config setting 'usb_power_temperature_offset' missing, please add it to config.py
2021-01-01 00:00:02 [warning  / 140kB] > config setting 'wifi_country' missing, please add it to config.py
2024-05-02 18:04:54 [info     / 138kB] > performing startup
2024-05-02 18:04:54 [debug    / 136kB]   - running Enviro 0.0.10, MicroPython v1.22.2, enviro v1.22.2 on 2024-03-06
2024-05-02 18:04:54 [info     / 129kB]   - wake reason: usb_powered
2024-05-02 18:04:54 [debug    / 127kB]   - turn on activity led
2024-05-02 18:04:54 [debug    / 122kB] > 90 blocks free out of 212
2024-05-02 18:04:54 [debug    / 120kB] > taking new reading
2024-05-02 18:04:54 [info     / 116kB]   - seconds since last reading: 19
2024-05-02 18:04:57 [debug    / 118kB] > caching reading for upload
2024-05-02 18:04:57 [info     / 108kB] > 25 cache file(s) need uploading
2024-05-02 18:04:57 [info     / 106kB] > connecting to wifi network 'WIFI'
2024-05-02 18:04:57 [info     / 102kB] > MAC: 28:cd:c1:05:1f:55
2024-05-02 18:04:57 [info     / 100kB] > active: 0, status: 0 (Link is down)
2024-05-02 18:04:57 [info     /  98kB] > Ready for connection!
2024-05-02 18:04:57 [info     / 141kB] > Connecting to SSID ASUS (password: *********)...
2024-05-02 18:04:58 [info     / 137kB] > active: 1, status: 1 (Connected to wifi)
2024-05-02 18:04:59 [info     / 133kB] > active: 1, status: 1 (Connected to wifi)
2024-05-02 18:04:59 [info     / 129kB] > active: 1, status: 1 (Connected to wifi)
2024-05-02 18:05:00 [info     / 125kB] > active: 1, status: 1 (Connected to wifi)
2024-05-02 18:05:00 [info     / 121kB] > active: 1, status: 1 (Connected to wifi)
2024-05-02 18:05:01 [info     / 117kB] > active: 1, status: 3 (Connect to wifi with an IP address)
2024-05-02 18:05:01 [info     / 114kB] > Connected successfully!
2024-05-02 18:05:01 [info     / 112kB] > IP: 192.168.1.4, Subnet: 255.255.255.0, Gateway: 192.168.1.1, DNS: 192.168.1.1
2024-05-02 18:05:01 [info     / 110kB] > Elapsed: 4131ms
2024-05-02 18:05:01 [info     / 136kB] > uploading cached readings to MQTT broker: 192.168.1.250:1883
2024-05-02 18:05:02 [debug    / 131kB]   - an exception occurred when uploading. Traceback (most recent call last):
  File "enviro/destinations/mqtt.py", line 28, in upload_reading
  File "enviro/mqttsimple.py", line 69, in connect
OSError: -2

2024-05-02 18:05:02 [debug    / 127kB]   - an exception occurred when disconnecting mqtt client. Traceback (most recent call last):
  File "enviro/destinations/mqtt.py", line 40, in upload_reading
  File "enviro/mqttsimple.py", line 116, in disconnect
OSError: [Errno 107] ENOTCONN

2024-05-02 18:05:02 [error    / 125kB]   ! failed to upload '2024-05-02T14_52_46Z.json' to mqtt
2024-05-02 18:05:02 [info     / 122kB] > Disconnecting wireless after upload
2024-05-02 18:05:02 [error    / 120kB] ! reading upload failed
2024-05-02 18:05:02 [info     / 118kB] > going to sleep
2024-05-02 18:05:02 [debug    / 116kB]   - clearing and disabling previous alarm
2024-05-02 18:05:02 [info     / 114kB]   - setting alarm to wake at 18:15pm
2024-05-02 18:05:02 [info     / 140kB]   - shutting down
2024-05-02 18:05:02 [debug    / 138kB]   - on usb power (so can't shutdown). Halt and wait for alarm or user reset instead

```

Would anyone know what I can do, I'm bumping into walls all over the place trying to get it to be recognised.

MrDrem, 2024-05-02:
Could you post your (suitably redacted) config.py file?

MazenS, 2024-05-02:
x

> Could you post your (suitably redacted) config.py file?

`# enviro config file

# you may edit this file by hand but if you enter provisioning mode
# then the file will be automatically overwritten with new details

provisioned = True

# enter a nickname for this board
nickname = 'grow'

# network access details
wifi_ssid = 'ASUS'
wifi_password = 'iamamoron'
wifi_country = "CZ"

# how often to wake up and take a reading (in minutes)
reading_frequency = 15

# where to upload to ("http", "mqtt", "adafruit_io", "influxdb")
destination = 'mqtt'

# how often to upload data (number of cached readings)
upload_frequency = 1

# web hook settings
custom_http_url = ''
custom_http_username = ''
custom_http_password = ''

# mqtt broker settings
mqtt_broker_address = '192.168.1.250:1883'
mqtt_broker_username = 'mqtt-grow'
mqtt_broker_password = 'alanmoore1'

# adafruit ui settings
adafruit_io_username = ''
adafruit_io_key = ''

# influxdb settings
influxdb_org = ''
influxdb_url = 'http://192.168.1.250:8086'
influxdb_token = ''
influxdb_bucket = 'homeassistant'

# grow specific settings
auto_water = False
moisture_target_a = 0
moisture_target_b = 0
moisture_target_c = 0
`

MazenS, 2024-05-03:
Of course - first time I didn't do the redacting properly

> Could you post your (suitably redacted) config.py file?

provisioned = True

nickname = 'grow'

wifi_ssid = 'WIFI'
wifi_password = 'PASSWORD'
wifi_country = "CZ"

reading_frequency = 15

destination = 'mqtt'

upload_frequency = 1

custom_http_url = ''
custom_http_username = ''
custom_http_password = ''

mqtt_broker_address = '192.168.1.250:1883'
mqtt_broker_username = 'mqtt-grow'
mqtt_broker_password = 'PASSWORD'

adafruit_io_username = ''
adafruit_io_key = ''

influxdb_org = ''
influxdb_url = 'http://192.168.1.250:8086'
influxdb_token = ''
influxdb_bucket = 'homeassistant'

auto_water = False
moisture_target_a = 0
moisture_target_b = 0
moisture_target_c = 0

MazenS, 2024-05-29:
> Could you post your (suitably redacted) config.py file?

Checking in again to see if you can offer any help or pointers?

mattroell, 2025-06-20:
Did you get this working for Home Assistant? It works for me.