pimoroni/enviro #227: Enviro weather MQTT with Mosuito on Hassio

davidbenucci, 2024-05-12:
I configured mosquito on Hassio and set up enviro weather to send readings to it (one readfing every 15 mins and upload after 10 readings). But I noticed on the graph are reported only the value  from the last reading of each group of json files. Do I have to configure somenthing more on mosquito? Thanks in advance?
I attache the grap.
[image]

CBDesignS, 2024-05-28:
I could be wrong here but it used to be that the data uploaded even tho it may have 10 sets of readings stored in it only has one timestamp (unless this has been fixed) so there is no way to split them up to get the individual readings out and the only reading that makes sense to ha is the last one with a timestamp. I could be wrong on this but from memory that is how it used to work,. Maybe somone else with more knowledge of the code can let you know if I am wrong.

ievren, 2024-07-15:
I think the way Home Assistant works, this is not possible?. See:
- https://github.com/home-assistant/home-assistant.io/issues/33742

@CBDesignS: Do you mean something like this: https://community.home-assistant.io/t/sensor-transmits-multiple-time-points-how-to-ingest-into-ha/527887?

Actually, when I tried the same thing as @davidbenucci, the config file on Enviro was:

```
# how often to wake up and take a reading (in minutes)
reading_frequency = 2

# how often to upload data (number of cached readings)
upload_frequency = 5
```

I see this in MQTT Explorer (https://mqtt-explorer.com/) in the history section—multiple messages, but each of them has its own timestamp. 

@CBDesignS  not sure what you are referring to with:
`it may have 10 sets of readings stored in it only has one timestamp (unless this has been fixed)`

Example from the "history" section:

Last message:
`07/15/2024 9:46:17 AM`
```
{
  "readings": {
    "pressure": 963.92,
    "pm1": 10,
    "pm2_5": 24,
    "noise": 0.046,
    "humidity": 55.7,
    "temperature": 25.69,
    "pm10": 24
  },
  "nickname": "enviro-urban",
  "model": "urban",
  "uid": "e66164084356202d",
  "timestamp": "2024-07-15T07:46:09Z"
}
```

2. 07/15/2024 9:46:16 AM (-0.51 seconds)
```
{
  "readings": {
    "pressure": 963.93,
    "pm1": 10,
    "pm2_5": 16,
    "noise": 0.05,
    "humidity": 55.89,
    "temperature": 25.66,
    "pm10": 30
  },
  "nickname": "enviro-urban",
  "model": "urban",
  "uid": "e66164084356202d",
  "timestamp": "2024-07-15T07:44:09Z"
}
```

3. 07/15/2024 9:46:16 AM (-0.51 seconds)
```
...
"timestamp": "2024-07-15T07:42:09Z"
...
```

4. 07/15/2024 9:46:16 AM (-0.51 seconds)
```
...
"timestamp": "2024-07-15T07:40:09Z"
...
```

5. 07/15/2024 9:46:15 AM (-0.3 seconds)
```
...
"timestamp": "2024-07-15T07:38:09Z"
...
```

6. 07/15/2024 9:36:14 AM (-10 minutes)
```
...
"timestamp": "2024-07-15T07:36:09Z"
...
```

CBDesignS, 2024-07-15:
I am not the person to help here as I actually gave up on the enviro weather board after bench testing for a month or two and it was put back in the box it arrived in. I already had an old but functioning FineOffset WHx80 setup that I was looking to replace with the Enviro Weather but the problems and lack of support at the time I bought the enviro made me decided to spend the time getting the old reliable setup working and I even added to the system with a lightening strike detector and Lux meter. They are stand alone 433mhz radio units that just send data to a 433mhz reciever that talks directly to HA and HA defined sensors and then HA uses the data to give me the graphs and readings I need, No software, hardware, firmware or support problems with them. I really wanted to use the Enviro Weather as it was newer and more compact.

Good luck getting your board and ha setup. 

this is what my weather station looks like in HA.
[image]